Lash-L/Ender-3-V2-Neo-Setup #1: Board version/NOZZLE_TO_PROBE_OFFSET in configuration.h

Perrotn, 2023-01-06:
Hi,

I am new to the 3d printing hobby.

Thank you for this tutorial, I did update my firmware successfully. I still need to try its feature though.

Could you clarify the part: Delete the Generic V4 and replace with the V4.2.2 or V4.2.7

Here are the lines 92-94 of the unmodified configuration.h file from https://github.com/mriscoc/Ender3V2S1
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_CREALITY_V4  // Creality Board v4.2.2
#endif

I tried modified it like bellow but the compilation failed.
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_CREALITY_V4.2.2  // Creality Board v4.2.2
#endif

I just compiled it with the raw value and I did not had an error, but as you specified that it need to be change I am wondering if you could help me with that.

Also here are my NOZZLE_TO_PROBE_OFFSET measurements:
#define NOZZLE_TO_PROBE_OFFSET { -39.5, -11, -1.47 } 
I pressed the nozzle on a notebook (it leave a mark where the nozzle where pressed) and then I just mark the X/Y position of the tip of the sensor. I measure the X/Y with a caliper and used your Z offset as I could find a way to measure it with accuracy (I will calibrate it with the z-offset). Could you share the way you measure the X/Y nozzle to probe offset?

Thank you

conway220, 2023-01-06:
Hey @Perrotn - Sorry for the delay in getting back to you, my internet has been offline for a few hours.
> Could you clarify the part: Delete the Generic V4 and replace with the V4.2.2 or V4.2.7

Yes! So if you look at my example [configuration ](https://github.com/conway220/Ender-3-V2-Neo-Setup/blob/main/Marlin_Configurations/Configuration.h) - You can see I have the following:

~~~
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_CREALITY_V4  
  // #define MOTHERBOARD BOARD_CREALITY_V422 //if you have 4.2.2
  // #define MOTHERBOARD BOARD_CREALITY_V427 //if you have 4.2.7
#endif
~~~

For a 4.2.2 board you should change it to CREALITY_V422. This whole step is relatively optional, as the generic V4 will work as well, but I figure it is better to be more specific when you can.

> Also here are my NOZZLE_TO_PROBE_OFFSET measurements:
#define NOZZLE_TO_PROBE_OFFSET { -39.5, -11, -1.47 }
I pressed the nozzle on a notebook (it leave a mark where the nozzle where pressed) and then I just mark the X/Y position of the tip of the sensor. I measure the X/Y with a caliper and used your Z offset as I could find a way to measure it with accuracy (I will calibrate it with the z-offset). Could you share the way you measure the X/Y nozzle to probe offset?

Teaching Tech has a good video on the entire process. You basically just use a caliper to measure between the nozzle to the probe. I typically don't bother doing any type of Z measurement until the firmware is installed as I find it difficult.

https://youtu.be/fN_ndWvXGBQ?t=177

Hope this helps! Let me know if you need anything else

Perrotn, 2023-01-06:
Thank you so much for your answer and the resources you shared.

conway220, 2023-01-06:
Of course, let me know if you have any other questions

closed 2023-01-06 by conway220