Lash-L/Ender-3-V2-Neo-Setup #3: firmware install

ImMrBalloon, 2023-02-01:
My firmware will not install to the printer. Anyone know what the issue may be? I think I did everything right.

ImMrBalloon, 2023-02-01:
when i put in the sd card the screen just shows the creality logo. I have tried with the just the .bin on the card as well and that does not work either

Lash-L, 2023-02-01:
Does it stay on the creality logo or does it eventually go to the main screen?

Have you tried renaming the .bin?

Are you sure you have your motherboard set to the correct version?

Is your sd card formatted correctly? Fat32 4096 allocation bytes?

ImMrBalloon, 2023-02-01:
The creality logo stays on and it doesn’t go to the main screen. I believe I have it set to the right mobo and will double check. Do I have to rename the .bin to anything special? I have renamed it a few times.

Lash-L, 2023-02-02:
Nope, you shouldn't need to rename it to anything special. Just a new name. How long do you wait on the Creality logo? do you have anything plugged into the printers micro usb?

Lash-L, 2023-02-13:
@ImMrBalloon If I don't hear from you before the end of the week, I'm going to close this issue, otherwise I'm happy to try to help

ImMrBalloon, 2023-02-13:
Oh, sorry I didn’t see your last reply! I got the firmware to work by renaming it to something shorter. Thanks for the help.

closed 2023-02-13 by ImMrBalloon