Lash-L/Ender-3-V2-Neo-Setup #4: Bed size - might not be an issue

praywood, 2023-02-06:
Hey Lash - Thanks for putting this together.  I just purchased my first 3d printer so all of this is all very new to me.  I apologize in advance if im not understanding your direction properly.  

I followed your firmware update instruction and downloaded the source code zip file from Dec 22, 2022 under mriscoc.  https://github.com/mriscoc/Ender3V2S1/archive/refs/tags/20221222.zip

I then downloaded your configuration for Marlin and overwrote the files outlined in steps two and three under step one of configuring Marlin.  Now, after watching some YouTube videos on using vscode I installed, auto build marlin in addition to platformio.  When the marlin auto build runs you get a projected output before compile, this projection shows a Cartesian 230x230x250mm heated bed, but I haven't seen an Ender 3 v2 neo with this large of a bed and after measuring mine its 220x220, is this an issue or are the 10 extra mm on each side supposed to be there? or shame on me for going rogue on your original work instruction by downloading Auto Build Marlin?

Lash-L, 2023-02-06:
It really shouldn't matter too much, as the size limits should be mainly handled by your slicer. But I got those sizes from here: https://github.com/MarlinFirmware/Configurations/blob/import-2.1.x/config/examples/Creality/Ender-3%20V2%20Neo/Configuration.h and here https://all3dp.com/2/ender-3-bed-size-what-is-it-really/#:~:text=Great%20Printers%2C%20Small%20Secrets,-We%20can%20go&text=All%20three%20of%20the%20Ender,print%20bed%20is%20going%20unused!

Both of which state 235mm. I have never actually measured it. But if you measured and your positive about your measurement at 220, then yeah I would keep it at 220. Creality advertised 220, but they do that for all of their ender 3s and typically, it is actually 235m. But it is very possible that the v2 neo is actually just 220. Give it one more measure and let me know what you find.

praywood, 2023-02-06:
I think you're right, it shouldn't matter.  I'm traveling this week so won't be able to double check till the weekend

Lash-L, 2023-02-13:
Were you able to double check @praywood ?