Lash-L/Ender-3-V2-Neo-Setup #6: Possible changes in firmware upgrade process?

prashnts, 2023-03-05:
Hi! Thanks for this detailed instruction and your config. I have successfully followed it to install updated firmware.

While installing, I noticed that the process after inserting SD card into the printer was slightly different. I inserted the SD card, turned on the printer, then instead of a black screen I had a Creality boot screen saying the firmware was being updated and wait for reboot. The reboot never happened but I was greeted by new UI on screen with "Ender3 (model) ready". Everything worked without reboot from that time on.

For the display firmware upgrade, I was able to take a picture. Here instead of blue/orange screen as you mention, there was a clear log output. I did have to reboot, remove SD card to use the printer again tho.

[image]

Lash-L, 2023-03-07:
@prashnts I wrote those settings from memory, so it is possible they have always been like that. Probably worth it for me to go through and double check/ add photos

prashnts, 2023-03-07:
Maybe, but actually I'd seen similar instructions elsewhere too (particularly in a youtube video). 

So it could be new firmare.elf from Creality maybe?